Chris--A/PrintEx #6: Example doesn't compile

dragospuri, 2015-09-13:
Using Arduino IDE 1.6.5
Installed the library through the manager. Opened the example. Doesn't compile.

Arduino: 1.6.5 (Windows 7), Board: "Arduino Nano, ATmega328"

Using library PrintEx in folder: C:\Users\Surreal\Documents\Arduino\libraries\PrintEx

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\eightanaloginputs -IC:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src C:\Users\Surreal\AppData\Local\Temp\build139473521994345729.tmp\basic_usage.cpp -o C:\Users\Surreal\AppData\Local\Temp\build139473521994345729.tmp\basic_usage.cpp.o

In file included from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintVariadic.h:8:0,
from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintExtension.h:26,
from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/MemoryPrinter.h:9,
from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/PrintEx.h:8,
from basic_usage.ino:8:
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:106:25: error: wrong number of template arguments (1, should be 2)
!is_pointer::value>::type
^
In file included from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintVariadic.h:8:0,
from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintExtension.h:26,
from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/MemoryPrinter.h:9,
from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/PrintEx.h:8,
from basic_usage.ino:8:
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:42:41: error: provided for 'template struct enable_if'
template< bool V, typename T > struct enable_if{}; //C++98 must provide all template parameters.
^
In file included from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintVariadic.h:8:0,
from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintExtension.h:26,
from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/MemoryPrinter.h:9,
from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/PrintEx.h:8,
from basic_usage.ino:8:
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:107:3: error: template argument 3 is invalid

{
^
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:107:4: error: expected '::' before '{' token
{
^
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:107:4: error: expected identifier before '{' token
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:107:4: error: qualified name does not name a class before '{' token
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:119:24: error: wrong number of template arguments (1, should be 2)
is_pointer::value>::type
^
In file included from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintVariadic.h:8:0,
from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintExtension.h:26,
from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/MemoryPrinter.h:9,
from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/PrintEx.h:8,
from basic_usage.ino:8:
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:42:41: error: provided for 'template struct enable_if'
template< bool V, typename T > struct enable_if{}; //C++98 must provide all template parameters.
^
In file included from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintVariadic.h:8:0,
from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintExtension.h:26,
from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/MemoryPrinter.h:9,
from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/PrintEx.h:8,
from basic_usage.ino:8:
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:120:3: error: template argument 3 is invalid
{ 
^
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:120:4: error: expected '::' before '{' token
{ 
^
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:120:4: error: expected identifier before '{' token
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:120:4: error: qualified name does not name a class before '{' token
In file included from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintExtension.h:26:0,
from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/MemoryPrinter.h:9,
from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/PrintEx.h:8,
from basic_usage.ino:8:
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintVariadic.h:19:34: warning: variadic templates only available with -std=c++11 or -std=gnu++11 [enabled by default]
template< typename T, typename... U >
^
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintVariadic.h:20:46: warning: variadic templates only available with -std=c++11 or -std=gnu++11 [enabled by default]
derived &printx( const T& t, const U&... u ){
^
In file included from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/PrintEx.h:9:0,
from basic_usage.ino:8:
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/StreamExtension.h:20:19: error: expected template-name before '<' token
ios::OStreamBase,
^
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/StreamExtension.h:20:19: error: expected '{' before '<' token
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/StreamExtension.h:20:19: error: expected unqualified-id before '<' token
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/StreamExtension.h:27:23: error: invalid use of incomplete type 'struct ios::StreamExtension'
struct StreamEx : StreamExtension{
^
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/StreamExtension.h:16:9: error: forward declaration of 'struct ios::StreamExtension'
struct StreamExtension : 
^
basic_usage.ino:48:14: error: expected '}' at end of input
expected '}' at end of input

Chris--A, 2015-09-13:
Hi, I'm having a quick look into it. I'm just downloading the latest IDE to confirm, however it looks like it is due to C++11 not being active in 1.6.5

I recall it was activated in 1.6.5r2 (current is 1.6.5r5).

Sorry I should have checked and added a note for the minimum version. I'll also fix the code so the C++11 only features are `#defined` to not cause error in the older IDE's (streaming is C++11 specific and parts of the code are not prevented from compiling).

Chris--A, 2015-09-13:
A small update, 1.6.5r5 does not include the changes pushed to the repo. I have opened an issue for this and will report back once worked out. (https://github.com/arduino/Arduino/issues/3791)

For now, I'm going through the code and I'll make it C++98 safe. I release the changes in version 1.1.6 today.

Thank you for taking the time to report this.

PS: if you would like to use the library now, you can enable C++11 by adding the changes found here to your platform.txt file: https://github.com/arduino/Arduino/commit/ddf4d878fdb39173b872f69149fb4cf1cd9fd017

Chris--A, 2015-09-14:
This is now fixed. Version 1.1.6 is available and should appear in the library manager within the hour.

As a result, in IDE versions below 1.6.6 (not released yet), Streaming and variadic printing are unavailable. You can however enable C++11 yourself to access these features.

Thank you for your input.

closed 2015-09-14 by Chris--A