Chris--A/PrintEx #23: Please change your license to LGPL?

dprophet, 2016-12-18:
Why are you using GNU GPL?

Please change your licensing to

GNU LGPL

https://www.gnu.org/licenses/lgpl-3.0.en.html

Your license means you must opensource everything, sketches and all if I use your printf. That just doesnt make any sense. GNU library license is what you are looking for.

dprophet, 2016-12-18:
No big deal. You use to have an MIT license.

I just forked,
git reset --hard 9c873d406db24f889b765ae6a96e2e3f6afa4e76
And pushed.

I will use an older version before you added Richard Stallman's GPL. 1 year old is fine for me as I dont need anything other than simple printf in a library extension.

Chris--A, 2016-12-20:
I can look at changing it. I just wanted to move away from the MIT licence.

There are some major bug fixes already in the code since the licence change,,  and about 30 commits in the dev branch waiting. Also for version 2.0 printf has some new additions and some of the printf options change a little bit to reflect other standard printf implementations.

dprophet, 2016-12-20:
Sorry. I cant use it. GPL is the worse license. If I make any IoT system and link against your library I have to open source everything. GPL is for applications, LGPL is for libraries.

Please change to LGPL if you want people to use it.

Chris--A, 2016-12-21:
> Please change to LGPL if you want people to use it.

That is what I'm saying I can do.
I will finish some more updates for it and I'll look at publishing version 2.0.0 as LGPL.

dprophet, 2016-12-22:
Thanks. I will watch your repo until you publish LGPL.

MCUdude, 2017-12-25:
Any updates on this? 👍

phrxmd, 2018-01-14:
It would be really useful to have this either LGPL'd, or a statement that you're finally going to LGPL it and stick with GPL (in which case people can fork your old MIT-licensed version)