Chris--A/PrintEx #32: PrintEx conflicts with another library on ESP32?

scottchiefbaker, 2019-03-22:
I have a project that compiles just fine on avr and ESP8266, but **not** on ESP32. When I compile my project on ESP32 I get an error about redefining `select`?

    /home/bakers/Arduino/libraries/PrintEx/src/lib/TypeTraits.h:78:55: error: 'template<bool V, class T, class U> struct select' redeclared as different kind of symbol
     template< bool V, typename T, typename U > struct select { typedef U type; };

[esp32-error.txt](https://github.com/Chris--A/PrintEx/files/2994586/esp32-error.txt)

It's either from one of the core ESP32 IP libraries, or the ESP32 WebServer library. I don't know how to dig any deeper as the error message is a little beyond me. I'm glad to help test troubleshoot if you can tell me where to start.

scottchiefbaker, 2019-03-22:
After some digging I was able to recreate the error with the small test case.

    #include <WebOTA.h>
    #include <PrintEx.h>

    // the setup function runs once when you press reset or power the board
    void setup() {
        Serial.begin(115200);
    }

    // the loop function runs over and over again forever
    void loop() {

    }

Sketch uses the [WebOTA](https://github.com/scottchiefbaker/ESP-WebOTA) library which I wrote. There is no mention of `select` in that code though. I don't understand what the error means so I'm not sure what the next step is.

scottchiefbaker, 2019-03-22:
Digging through the trace it looks like an include of an include of an include... Looks like maybe it's a conflict with this:

https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/newlib/sys/select.h#L25

`PrintEx` declares `select` as a template, but this library has already declared it as an int?

slundell, 2019-08-30:
As a temporary workaround you can wrap the include in its own namespace:

```
namespace printex {
#include <PrintEx.h>
}
printex::StreamEx serial = Serial;
```

scottchiefbaker, 2019-08-30:
I'm guessing at this point this library has been abandonned? Last commit was from 2016.

slundell, 2019-08-30:
Also, I was wrong. Fails to link :-\

slundell, 2019-08-31:
Seems like just commenting out the offending lines work fine: https://github.com/livello/PrintEx/commit/b430e1eb3a7a334d32464c669cf9b3df4c826c15